jp112sdl/MatrixDisplayESP8266 #3: Einbindung Iobroker

mikewolf, 2018-03-23:
Hallo , gibt es  irgendwo ne Anleitung wie ich das Display via Iobroker per Mqtt beschreiben kann ?
kann leider nichts finden und die CCU möchte ich eigentlich nicht befeuern , diese soll sich nur um Wichtige Dinge kümmern ,alles andere nice to have via Iobroker .

Mit freundlichen Grüßen
mikewolf

mikewolf, 2018-03-24:
Super, Danke werde ich heute noch probieren,
eine Frage hätte ich noch 👍 
kann man nachträglich noch die Einstellungen ändern ,oder geht die Änderung nur bei Ersteinrichtung,
habe diesbezüglich nichts im Sketch gefunden ?
mfg
mikewolf

jp112sdl, 2018-03-24:
Änderung geht nur via Ersteinrichtung.
So oft ändert man da ja auch nix ;)

mikewolf, 2018-03-25:
perfekt Danke !

closed 2018-03-25 by mikewolf