jp112sdl/MatrixDisplayESP8266 #5: Kriege es nicht zum Laufen - Anzeige spinnt...

lalam123, 2018-04-03:
Hallo Jerome,

vielleicht kannst Du bitte helfen? 
Bauteile wie Einkaufsliste.
Alles laut Anleitung (allerdings erstmal nur mit einem Display) aufgebaut, WiFi konfiguriert, HTTP-Ausgabe kontrolliert im Quelltext.- Passt alles. WiFi des ESP steht.
Das LED-Modul ist dauerhaft voll beleuchtet, flackert jedoch periodisch als ob es was anzeigen will. Habe In der MD_Max72xx.h auf  USE_FC16_HW gestellt und auch alle anderen durchprobiert. Keine nennenswerten Änderungen, allenfalls eine andere Nuance beim Flackern :-)  #define MAX_DEVICES steht auf 4.

Muss ich noch etwas anderes im Sketch anpassen? Kompiliert sauber ohne Fehler durch. 

Hast Du da noch einen Tip für mich?

Danke und Liebe Grüße
Lars

jp112sdl, 2018-04-03:
Moin!

Hast du mal ein Example-Sketch aus der MD_Parola Library getestet, ob das korrekt angezeigt wird?

Wenn das läuft, jedoch der Sketch von mir nicht, dann kannst du mal in [Zeile 28](https://github.com/jp112sdl/MatrixDisplayESP8266/blob/6d62e62cc6433938bad530d6449726b97e99c7d5/MatrixDisplayESP8266.ino#L28) die Kommentarzeichen (`//`) entfernen und sie vor Zeile 29 schreiben.

```
MD_Parola P = MD_Parola(DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);
//MD_Parola P = MD_Parola(CS_PIN, MAX_DEVICES);
```

lalam123, 2018-04-03:
Oh das geht ja schnell :-)
Ich probiere ...

lalam123, 2018-04-03:
Hab das Hello World Beispiel mal geladen, alles leuchtet voll. Habe ich die PINS richtig?

#define CLK_PIN   D7 oder #define CLK_PIN   7 ?
#define DATA_PIN  D5
#define CS_PIN    D8

jp112sdl, 2018-04-03:
`D7` ist  richtig, oder du schreibst `13`. Das ergibt sich aus dem Wemos-Arduino-Pin-Mapping.
Aber ich sehe dabei gerade was anderes...
Standard-CLK Pin ist `D5` und Standard-MOSI, also Daten ist `D7`.

So ist es auch in einem MD_Parola Beispiel definiert:
```
#define CLK_PIN   D5 // or SCK
#define DATA_PIN  D7 // or MOSI
```

Also tausch mal deine Drähte D5<>D7 und versuche noch mal ein MD_Parola Beispiel zum Laufen zu bekommen. Bzw. ändere in meinem Sketch dann auch mal CLK auf D5 und DATA auf D7

lalam123, 2018-04-03:
Teilerfolg, das Hello Script läuft ... :-))
.... melde mich wieder ....

lalam123, 2018-04-03:
Es läuft !!!!!!
Vielen lieben Dank - da haste wieder jemanden glücklich gemacht!
Danke, Danke, Danke !!!!!!

jp112sdl, 2018-04-03:
Wenn du mir jetzt noch verrätst, in welcher Konstellation alles läuft... :)

Also an welche Wemos-Pins hast du jetzt CLK und DATA vom Matrixdisplay angeschlossen?

Und welcher Code funktioniert zur Initialisierung?
`MD_Parola P = MD_Parola(DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);`
oder
`MD_Parola P = MD_Parola(CS_PIN, MAX_DEVICES);`

lalam123, 2018-04-03:
Habe nur die Kabel vertauscht, Da hab ich mich wohl beim Verdrahten vertan. 
CLK läuft an D5 und Data an D7. Code ist Original geblieben.

jp112sdl, 2018-04-03:
Alles klar. Hab den Code trotzdem mal an die native Pin-Nutzung angepasst.
<hr>
Wenns jetzt bei dir läuft, dann mach ich hier zu :)

closed 2018-04-03 by jp112sdl