jp112sdl/MatrixDisplayESP8266 #9: Fehler beim Sketch überprüfen

Michel83, 2018-04-15:
Hi, ich habe ein Problem wenn ich versuche den Sketch zu überprüfen und zwar bekomm ich folgende Meldung:

C:\Users\Michael\Documents\Arduino\libraries\MD_Parola\src\MD_Parola_Sprite.cpp: In member function 'uint8_t* MD_PZone::setupSprite(uint8_t, uint8_t&, uint8_t&)':

C:\Users\Michael\Documents\Arduino\libraries\MD_Parola\src\MD_Parola_Sprite.cpp:65:29: error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]

   case SPR_PACMAN:        p = pacman1; dataWidth = W_PMAN1; numFrames = F_PMAN1; break;

                             ^

C:\Users\Michael\Documents\Arduino\libraries\MD_Parola\src\MD_Parola_Sprite.cpp:66:29: error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]

   case SPR_PACMAN_GHOST:  p = pacman2; dataWidth = W_PMAN2; numFrames = F_PMAN2; break;

                             ^

C:\Users\Michael\Documents\Arduino\libraries\MD_Parola\src\MD_Parola_Sprite.cpp:67:29: error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]

   case SPR_ROCKET:        p = rocket;  dataWidth = W_ROCKET; numFrames = F_ROCKET; break;

                             ^

Mehrere Bibliotheken wurden für "WiFiUdp.h" gefunden
 Benutzt: C:\Users\Michael\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi
 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\WiFi
 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\WiFi
 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\WiFi
 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Fehler beim Kompilieren für das Board WeMos D1 R2 & mini.
Ungültige Bibliothek C:\Users\Michael\Documents\Arduino\libraries\MatrixDisplayESP8266-master in C:\Users\Michael\Documents\Arduino\libraries\MatrixDisplayESP8266-master gefunden`` 

Ich hab keine Ahnung was ich Falsch gemacht habe

Michel83, 2018-04-15:
Ich habe es gefunden es war diese Liberie MD_Parola mit der Version 2.7.0 Funktioniert es

jp112sdl, 2018-04-15:
Ok. Ich konnte es auch nicht nachvollziehen / reproduzieren :)

closed 2018-04-15 by jp112sdl