jp112sdl/MatrixDisplayESP8266 #12: Aussehen Problem

Dufy111, 2018-04-21:
Hallo
Was ist das Problem?
Die Uhr ist turbulent!
Video-Link: https://www.youtube.com/watch?v=i00DJQdwcFc

jp112sdl, 2018-04-21:
Falsches Device initialisiert.

Siehe auf
https://www.schellenberger.biz/matrix-display-fuer-homematic-im-nachbau/

Unteres Drittel der Seite bei
**2) Schrift spiegelverkehrt?**

Dort hat Micha erklärt, was zu tun ist.

Dufy111, 2018-04-21:
Danke für die Hilfe
Problem #define USE_FC16_HW 0 wurde auf 1 umgeschrieben
mit diesem Modul: https://www.aliexpress.com/item/1pcs-SAMIORE-ROBOT-MAX7219-dot-matrix-module-microcontroller-module-4-in-one-display/32689479860.html?spm=a2g0s.9042311.0.0.TQRJo7
jetzt perfekt!
Danke

jp112sdl, 2018-04-21:
Ist immer ein Glücksspiel... Man weiß nie, was man bekommt und muss immer ein bisschen mit der Lib rumspielen :)

Schön, dass es jetzt geht! 👍

closed 2018-04-21 by jp112sdl