jp112sdl/MatrixDisplayESP8266 #13: Geistervariable - Alle 3-12 Stunden zwei mal Temperatur

capitaenz, 2018-04-22:
Hallo Jérôme,

ich habe das Problem, dass mir das Display immer zwei Mal die Temperatur anzeigt.
Einmal die Aktuelle und einmal eine die "Hängengeblieben" ist.
Es handelt sich um die letzte Variable vor der Uhrzeit.
Die Raspberrymatic stellt die aktuellen und richtigen Variablen zur Verfügung, das habe ich überprüft.
Sobald ich Reset drücke am ESP8266, wird wieder alles für die nächsten Stunden richtig angezeigt.

Hier mein Script:
```
! ----------
! 8x8 ESP8266 Script
! ----------
var geschirr = dom.GetObject("Geschirr fertig").Value();
var waschmaschine = dom.GetObject("Wäsche fertig").Value();
var sound = dom.GetObject("Sound an/aus").Value();
var temp = dom.GetObject("HVL.Wunderground:1.TEMPERATURE").State();
var box = dom.GetObject("Staubbehaelter").Value();
var strom = dom.GetObject("Strom Verbrauch heute").Value();
var stromk = dom.GetObject("Strom Verbrauch heute").Value();
var watt = dom.GetObject("Strom Verbrauch aktuell").Value();
var wasser = dom.GetObject("Wasser Verbrauch heute").Value();
var DC = dom.GetObject("Status_DutyCycle").State();

string list;

! ----------
! Geschirr
! ----------

if (geschirr == true){
list = list + "Geschirr ist fertig;";
dom.GetObject('MAX_8x8').State(list);
}

! ----------
! Waschmaschine
! ----------

if (waschmaschine == true){
list = list + "Die Waschmaschine ist fertig;";
dom.GetObject('MAX_8x8').State(list);
}

! ----------
! Sound
! ----------
if (sound == false){
list = list + "Sound ist aus;";
dom.GetObject('MAX_8x8').State(list);
}

! ----------
! Staubbehälter
! ----------
if (box == true){
list = list + "Bitte noch die Staubbox vom HomBot leeren;";
dom.GetObject('MAX_8x8').State(list);
}

! ----------
! Stromverbrauch heute
! ----------

list = list + "Stromverbrauch:   ";
strom = strom.ToString(2);
list = list + strom;
list = list + " kWh;";

! ----------
! Stromkosten heute
! ----------

var skosten;
list = list + "Stromkosten:  ";
skosten = stromk * 0.2595;
skosten = skosten.ToString(2);
list = list + skosten;
list = list + " Euro;";

! ----------
! Stromverbrauch aktuell
! ----------
watt = watt * 1000;
watt = watt.ToString(0);
list = list + "Aktueller Stromverbrauch:  ";
list = list + watt;
list = list + " Watt;";

! ----------
! Wasserverbrauch heute
! ----------
var liter;
liter = wasser * 1000;
list = list + "Wasserverbrauch:  ";
list = list + liter.ToString(0);
list = list + " Liter;";

! ----------
! Wasserkosten
! ----------
list = list + "Wasserkosten:  ";
var wpreis;
wpreis = 0.004;
integer wkosten = (wasser * 1000) * wpreis;
list = list + wkosten.ToString(2);
list = list + " Euro;";

! ----------
! Duty Cycle
! ----------

list = list + "Duty Cycle:";
DC = DC.ToString(0);
list = list + DC;
list = list + "%;";

! ----------
!Temperatur
! ----------
list = list + "Auàentemperatur:        ";
temp = temp.ToString(1);
temp = temp + "$C";
list = list + temp;

dom.GetObject('MAX_8x8').State(list);

! ----------
!Debug
! ----------
!var esp = dom.GetObject("MAX_8x8").State();

!WriteLine(geschirr);
!WriteLine(waschmaschine);
!WriteLine(sound);
!WriteLine(temp);
!WriteLine(box);
!WriteLine(esp);
```

Hast du eine Idee warum dies so ist?

VG
Leif

jp112sdl, 2018-04-22:
Hallo Leif,

du hast dir da ein richtig schickes Skript gebaut!

Aber was zu deinem beschriebenen Verhalten führen könnte, entzieht jeglicher Ahnung.
Da hilft eigentlich nur, den Wemos mal seriell am PC mitloggen (115200 baud) und dann mal im Log zu schauen, ob irgendwas komisches kommt, in dem Moment, wo die Temperatur wieder 2x auftaucht.

capitaenz, 2018-04-23:
Moin Jérôme,

kannst du mir noch etwas mit auf den Weg geben? Einen link oder dergleichen? Ich bin nicht auf den Kopf gefallen, aber zu breit aufgestellt und ich müsste mich erst etwas einlesen, bevor ich dies machen könnte. Läuft dies ähnlich ab wie beim Arduino?

VG
Leif

P.S.: Danke für das Projekt! Gibt es ein "Spendenkonto"?

jp112sdl, 2018-04-23:
Läuft genau so wie beim Arduino...
Nimmst dir Putty, stellst auf Serial, 115200 Baud, COM-Port natürlich der vom angeschlossenen Wemos und dann auf "open".
Putty kann auch in eine Datei loggen.
Einen Link hab ich nicht... Ich arbeite auf dem Mac mit anderen Tools.

Aber notfalls geht auch der serielle Monitor aus der Arduino IDE.

Hier ist ein [Spendenbutton](https://www.paypal.com/cgi-bin/webscr?cmd=_s-xclick&hosted_button_id=UBX8NFNYVWW8N). :) 

Danke!!!

capitaenz, 2018-05-04:
Moin Jérôme,

ich bin noch nicht dazu gekommen den passenden Moment im Fehlerfall abzufangen, aber:
Könnte es mit einem instabilen WLan zusammenhängen? In unserer Häuserreihe gibt es Probleme, die sich in vielen verlorenen Netzwerkpaketen äußern und manchmal können Wlan System erst nach vielen Versuchen eine Verbindung aufbauen...

VG
Leif

jp112sdl, 2018-05-04:
Möglich ist sowas vielleicht... 
Mehr kann ich dir dazu aber auch nicht sagen, da ich 
- das geschilderte Verhalten nicht reproduzieren kann und 
- kein Log habe, in dem sowas _möglicherweise_ protokolliert sein könnte

capitaenz, 2018-10-12:
Hallo zusammen,

ich habe mir mit der Geistervariable jetzt geholfen, indem ich als letztes immer ein " " anzeigen lasse.
Dableibt zwar das Display kurz dunkel, aber besser als wenn die falsche Temperatur oder etwas anderes nicht mehr aktuelles angezeigt wird.
Es betrifft immer die letzte Anzeige...
Mein Script sieht jetzt so aus:

! ----------
! 8x8 ESP8266 Script
! ----------
var geschirr = dom.GetObject("Geschirr fertig").Value();
var waschmaschine = dom.GetObject("Wäsche fertig").Value();
var sound = dom.GetObject("Sound an/aus").Value();
var temp = dom.GetObject("HVL.Wunderground:1.TEMPERATURE").State();
var box = dom.GetObject("Staubbehaelter").Value();
var strom = dom.GetObject("Strom Verbrauch heute").Value();
var stromk = dom.GetObject("Strom Verbrauch heute").Value();
var watt = dom.GetObject("Strom Verbrauch aktuell").Value();
var wasser = dom.GetObject("Wasser Verbrauch heute").Value();
var DC = dom.GetObject("DutyCycle").State();
var PM25 = dom.GetObject("Feinstaub_PM2_5").Value();
var PM10 = dom.GetObject("Feinstaub_PM10").Value();

string list;

! ----------
! Geschirr
! ----------

if (geschirr == true){
list = list + "Geschirr ist fertig;";
dom.GetObject('MAX_8x8').State(list);
}

! ----------
! Waschmaschine
! ----------

if (waschmaschine == true){
list = list + "Die Waschmaschine ist fertig;";
dom.GetObject('MAX_8x8').State(list);
}

! ----------
! Sound
! ----------
if (sound == false){
list = list + "Sound ist aus;";
dom.GetObject('MAX_8x8').State(list);
}

! ----------
! Staubbehälter
! ----------
if (box == true){
list = list + "Bitte noch die Staubbox des HomBot leeren;";
dom.GetObject('MAX_8x8').State(list);
}

! ----------
! Stromverbrauch heute
! ----------

list = list + "Stromverbrauch:   ";
strom = strom.ToString(2);
list = list + strom;
list = list + " kWh;";

! ----------
! Stromkosten heute
! ----------

var skosten;
list = list + "Stromkosten:  ";
skosten = stromk * 0.2595;
skosten = skosten.ToString(2);
list = list + skosten;
 if (skosten >= 1) {list = list +" Euro;";}
 else {list = list +" cent;";}

! ----------
! Stromverbrauch aktuell
! ----------
watt = watt * 1000;
watt = watt.ToString(0);
list = list + "Aktueller Stromverbrauch:   ";
list = list + watt;
list = list + " Watt;";

! ----------
! Wasserverbrauch heute
! ----------
var liter;
liter = wasser * 1000;
list = list + "Wasserverbrauch:  ";
list = list + liter.ToString(0);
list = list + " Liter;";

! ----------
! Wasserkosten
! ----------
list = list + "Wasserkosten:  ";
var wpreis;
wpreis = 0.004;
integer wkosten = (wasser * 1000) * wpreis;
list = list + wkosten.ToString(2);
 if (wkosten >= 1) {list = list +" Euro;";}
 else {list = list +" cent;";}

! ----------
!Temperatur
! ----------
list = list + "Auàentemperatur:        ";
temp = temp.ToString(1);
temp = temp + "$C";
list = list + temp;
list = list + ";";

! ----------
! Duty Cycle
! ----------

list = list + "Duty Cycle:";
DC = DC.ToString(0);
list = list + DC;
list = list + "%;";

! ----------
! Feinstaub
! ----------

!list = list + "Feinstaubwerte:";
!PM25 = PM25.ToString(2);
!list = list + "PM2.5=";
!list = list + PM25;
!list = list + " - ";
!PM10 = PM10.ToString(2);
!list = list + "PM10=";
!list = list + PM10;
!list = list + ";";

! ----------
! Abschluss & gegen Geister
! ----------

list = list + " ";
dom.GetObject('MAX_8x8').State(list);

! ----------
!Debug
! ----------
var esp = dom.GetObject("MAX_8x8").State();

!WriteLine(geschirr);
!WriteLine(waschmaschine);
!WriteLine(sound);
!WriteLine(temp);
!WriteLine(box);
!WriteLine(strom);
!WriteLine(esp);
!WriteLine(PM25);

VG
Leif

closed 2018-12-25 by jp112sdl