jp112sdl/MatrixDisplayESP8266 #14: Zeitzone einstellen?

gerassimo, 2018-07-21:
Hallo Jérôme, 
ich betreibe das Display in Griechenland, läüft super!. Leider finde ich keinen Befehl, wo ich die Zeitzone einstellen kann (TimeLib.h/Time.h o.ä.). Kannst Du mir einen Tip geben?
Danke, gerassimo

jp112sdl, 2018-07-21:
Hi!
Ändere es am einfachsten in der `calcTime(time_t t)` ([TimeHandling.ino Zeile 39](https://github.com/jp112sdl/MatrixDisplayESP8266/blob/a344fc335026e57147b78add99773f70e6222dbb/TimeHandling.ino#L39)):
`byte Stunde = hour(t) + 1;` -> dort wird derzeit `+1` = 1 Stunde zur GMT addiert.

Wenn du also GMT+2 benötigst, machst da `byte Stunde = hour(t) + 2;` draus.

Die zusätzliche Stunde für die Sommerzeit wird dann wie gehabt noch addiert in [Zeile 40](https://github.com/jp112sdl/MatrixDisplayESP8266/blob/a344fc335026e57147b78add99773f70e6222dbb/TimeHandling.ino#L40)

Grüße!

gerassimo, 2018-07-21:
Hallo Jérôme,
danke, perfekt! 
lg gerassimo

closed 2018-07-21 by jp112sdl