jp112sdl/MatrixDisplayESP8266 #15: Kimpilier-Fehler

Seqway, 2018-08-19:
Hallo,
leider bekomme ich einen Kompilierfehler. Bibliotheken sind eingebunden, aber ob die richtige version eingestellt ist weiss ich nicht. 
Alle *.ino Dateien befinden sich im gleichen Verzeichnis. Was ist der Fehler ?

`Arduino: 1.8.5 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Build-Optionen wurden verändert, alles wird neu kompiliert
C:\Users\Entwicklung\Documents\Arduino\libraries\MD_Parola\src\MD_Parola.cpp: In constructor 'MD_Parola::MD_Parola(uint8_t, uint8_t, uint8_t, uint8_t)':

C:\Users\Entwicklung\Documents\Arduino\libraries\MD_Parola\src\MD_Parola.cpp:32:63: error: invalid conversion from 'uint8_t {aka unsigned char}' to 'MD_MAX72XX::moduleType_t' [-fpermissive]

 _D(dataPin, clkPin, csPin, numDevices), _numModules(numDevices)

                                                               ^

In file included from C:\Users\Entwicklung\Documents\Arduino\libraries\MD_Parola\src/MD_Parola.h:340:0,

                 from C:\Users\Entwicklung\Documents\Arduino\libraries\MD_Parola\src\MD_Parola.cpp:23:

C:\Users\Entwicklung\Documents\Arduino\libraries\MD_MAX72XX\src/MD_MAX72xx.h:345:3: error:   initializing argument 1 of 'MD_MAX72XX::MD_MAX72XX(MD_MAX72XX::moduleType_t, uint8_t, uint8_t, uint8_t, uint8_t)' [-fpermissive]

   MD_MAX72XX(moduleType_t mod, uint8_t dataPin, uint8_t clkPin, uint8_t csPin, uint8_t numDevices=1);

   ^

C:\Users\Entwicklung\Documents\Arduino\libraries\MD_Parola\src\MD_Parola.cpp: In constructor 'MD_Parola::MD_Parola(uint8_t, uint8_t)':

C:\Users\Entwicklung\Documents\Arduino\libraries\MD_Parola\src\MD_Parola.cpp:37:46: error: invalid conversion from 'uint8_t {aka unsigned char}' to 'MD_MAX72XX::moduleType_t' [-fpermissive]

 _D(csPin, numDevices), _numModules(numDevices)

                                              ^

In file included from C:\Users\Entwicklung\Documents\Arduino\libraries\MD_Parola\src/MD_Parola.h:340:0,

                 from C:\Users\Entwicklung\Documents\Arduino\libraries\MD_Parola\src\MD_Parola.cpp:23:

C:\Users\Entwicklung\Documents\Arduino\libraries\MD_MAX72XX\src/MD_MAX72xx.h:362:3: error:   initializing argument 1 of 'MD_MAX72XX::MD_MAX72XX(MD_MAX72XX::moduleType_t, uint8_t, uint8_t)' [-fpermissive]

   MD_MAX72XX(moduleType_t mod, uint8_t csPin, uint8_t numDevices=1);

   ^

Mehrere Bibliotheken wurden für "WiFiUdp.h" gefunden
 Benutzt: C:\Users\Entwicklung\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi
 Nicht benutzt: C:\Users\Entwicklung\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\WiFi
 Nicht benutzt: C:\Users\Entwicklung\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\WiFi
 Nicht benutzt: C:\Users\Entwicklung\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\WiFi
 Nicht benutzt: C:\Users\Entwicklung\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\WiFi
exit status 1
Fehler beim Kompilieren für das Board LOLIN(WEMOS) D1 R2 & mini.

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
`

jp112sdl, 2018-08-19:
Aufgrund von Änderungen in der MD_Parola neuester Version, funktionierte der Sketch nicht mehr.
Bitte mal mit dem [Sketch aus dem dev-Branch](https://github.com/jp112sdl/MatrixDisplayESP8266/blob/dev/MatrixDisplayESP8266.ino) testen.

Seqway, 2018-08-19:
Aha okay, werde ich mal versuchen. und welche Version von der MD_MAX72 sowie time ?

jp112sdl, 2018-08-19:
Die aktuelle... Hab da auch ewig nichts aktualisiert.
MD_MAX72XX ist auch die 3.0.1 bei mir und Time von Paul Stoffregen ausm Repo https://github.com/PaulStoffregen/Time

Seqway, 2018-08-19:
Ne mit der 3.0.1 kommt wieder der Fehler

In file included from C:\Users\Entwicklung\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\MatrixDisplayESP8266-master\MatrixDisplayESP8266-master\MatrixDisplayESP8266\MatrixDisplayESP8266.ino:1:0:

C:\Users\Entwicklung\Documents\Arduino\libraries\arduino_391117\src/MD_Parola.h:1081:38: error: expected ')' before 'mod'

   MD_Parola(MD_MAX72XX::moduleType_t mod, uint8_t dataPin, uint8_t clkPin, uint8_t csPin, uint8_t numDevices = 1);

Seqway, 2018-08-19:
Habe alle Bibliotheken und nochmal alles neu eingebunden und kompiliert nun kommt diese hier:

"C:\Users\Entwicklung\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Entwicklung\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/include" "-IC:\Users\Entwicklung\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include" "-IC:\Users\Entwicklung\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\ENTWIC~1\AppData\Local\Temp\arduino_build_946127/core" -c -w  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"   -DESP8266 "-IC:\Users\Entwicklung\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266" "-IC:\Users\Entwicklung\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\variants\d1_mini" "-IC:\Users\Entwicklung\Documents\Arduino\libraries\MD_Parola\src" "-IC:\Users\Entwicklung\Documents\Arduino\libraries\MD_MAX72XX\src" "-IC:\Users\Entwicklung\Documents\Arduino\libraries\Time-master" "-IC:\Users\Entwicklung\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src" "-IC:\Users\Entwicklung\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src" "-IC:\Users\Entwicklung\Documents\Arduino\libraries\ArduinoJson\src" "-IC:\Users\Entwicklung\Documents\Arduino\libraries\WiFiManager" "-IC:\Users\Entwicklung\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src" "-IC:\Users\Entwicklung\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\DNSServer\src" "-IC:\Users\Entwicklung\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ArduinoOTA" "-IC:\Users\Entwicklung\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\SPI" "-IC:\Users\Entwicklung\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266mDNS" "C:\Users\ENTWIC~1\AppData\Local\Temp\arduino_build_946127\sketch\MatrixDisplayESP8266.ino.cpp" -o "C:\Users\ENTWIC~1\AppData\Local\Temp\arduino_build_946127\sketch\MatrixDisplayESP8266.ino.cpp.o"
MatrixDisplayESP8266:28: error: invalid conversion from 'uint8_t {aka unsigned char}' to 'MD_MAX72XX::moduleType_t' [-fpermissive]

 MD_Parola P = MD_Parola(DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);

                                                               ^

In file included from C:\Users\Entwicklung\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\MatrixDisplayESP8266-master\MatrixDisplayESP8266-master\MatrixDisplayESP8266\MatrixDisplayESP8266.ino:1:0:

C:\Users\Entwicklung\Documents\Arduino\libraries\MD_Parola\src/MD_Parola.h:1081:3: error:   initializing argument 1 of 'MD_Parola::MD_Parola(MD_MAX72XX::moduleType_t, uint8_t, uint8_t, uint8_t, uint8_t)' [-fpermissive]

   MD_Parola(MD_MAX72XX::moduleType_t mod, uint8_t dataPin, uint8_t clkPin, uint8_t csPin, uint8_t numDevices = 1);

   ^

jp112sdl, 2018-08-19:
Diese Zeile
`MD_Parola P = MD_Parola(DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES); `
ist noch aus dem alten Sketch.

Wenn du meinem [Link von vorhin](https://github.com/jp112sdl/MatrixDisplayESP8266/issues/15#issuecomment-414139196) folgst, steht im Sketch:
`MD_Parola P = MD_Parola(HARDWARE_TYPE, DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);`

Seqway, 2018-08-19:
ja das habe ich in der Tat verbaselt. Hatte dies nun eingelesen und nun kommen JSON Fehler:

C:\Users\Entwicklung\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\MatrixDisplayESP8266-master\MatrixDisplayESP8266-master\MatrixDisplayESP8266\ConfigFileHandling.ino: In function 'bool loadSysConfig()':

ConfigFileHandling:17: error: 'DynamicJsonDocument' was not declared in this scope

   DynamicJsonDocument doc;

   ^

ConfigFileHandling:17: error: expected ';' before 'doc'

   DynamicJsonDocument doc;

                       ^

ConfigFileHandling:18: error: 'DeserializationError' was not declared in this scope

   DeserializationError error = deserializeJson(doc, buf.get());

   ^

ConfigFileHandling:18: error: expected ';' before 'error'

   DeserializationError error = deserializeJson(doc, buf.get());

                        ^

ConfigFileHandling:20: error: 'error' was not declared in this scope

   if (error) {

       ^

ConfigFileHandling:24: error: 'doc' was not declared in this scope

   JsonObject json = doc.as<JsonObject>();

                     ^

ConfigFileHandling:24: error: expected primary-expression before '>' token

   JsonObject json = doc.as<JsonObject>();

                                      ^

ConfigFileHandling:24: error: expected primary-expression before ')' token

   JsonObject json = doc.as<JsonObject>();

                                        ^

ConfigFileHandling:26: error: 'serializeJson' was not declared in this scope

   serializeJson(doc, Serial);

                            ^

C:\Users\Entwicklung\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\MatrixDisplayESP8266-master\MatrixDisplayESP8266-master\MatrixDisplayESP8266\ConfigFileHandling.ino: In function 'bool saveSysConfig()':

ConfigFileHandling:45: error: 'DynamicJsonDocument' was not declared in this scope

   DynamicJsonDocument doc;

   ^

ConfigFileHandling:45: error: expected ';' before 'doc'

   DynamicJsonDocument doc;

                       ^

ConfigFileHandling:46: error: 'doc' was not declared in this scope

   JsonObject json = doc.to<JsonObject>();

                     ^

ConfigFileHandling:46: error: expected primary-expression before '>' token

   JsonObject json = doc.to<JsonObject>();

                                      ^

ConfigFileHandling:46: error: expected primary-expression before ')' token

   JsonObject json = doc.to<JsonObject>();

                                        ^

ConfigFileHandling:62: error: 'serializeJson' was not declared in this scope

   serializeJson(doc, Serial);

jp112sdl, 2018-08-19:
Du hast jetzt nicht nur die `MatrixDisplayESP8266.ino` genommen, sondern auch die `ConfigFileHandling.ino` (diese funktioniert nur mit der ArduinoJSON 6.2.3-beta). 

Bitte ausschließlich die `MatrixDisplayESP8266.ino` ersetzen und die `ConfigFileHandling.ino` weiterhin aus dem master-Branch nutzen.

Seqway, 2018-08-19:
Okay, nun geht es mit dem masterfile für die ConfigFileHandling.ino !
Ich flashe das mal auf den Wemos jetzt.

Weisst du warum ich den Wemos, obwohl in meinem Netzwerk, ihn per ip Adresse im Browser nicht erreichen kann ?

jp112sdl, 2018-08-19:
Was möchtest du da erreichen?
Du kannst lediglich den Config-Mode starten, dich mit dem AP verbinden und dann deine Daten über http://192.168.4.1 eingeben

Seqway, 2018-08-19:
Wie starte ich denn den config Modus ? 
Ich habe allerdings auch noch nciht das Display dran. ich dachte aber ich komme auf diese Seite nochmal hier --> https://raw.githubusercontent.com/jp112sdl/MatrixDisplayESP8266/master/images/Konfiguration2.png

damit ich die Datenquelle noch einmal eingeben kann. Da soll ja später Datenpunkte per Laufschrift eingeblendet werden.

jp112sdl, 2018-08-19:
https://github.com/jp112sdl/MatrixDisplayESP8266#taster---funktion
> ein Gedrückthalten der Taste 1 oder der Taste 2 bei Einschalten/Stromzufuhr startet das Modul in den Konfigurationsmodus

Eine andere Möglichkeit, die URL zu ändern, ist nicht vorgesehen.

Seqway, 2018-08-19:
Ja das habe ich mehrfach gemacht. Taste 1 (habe die kleine Version vom Wemos) gedrückt halten und Strom anschliessen. Aber nichts passiert :-( In den Modus komme ich nicht mehr rein um die Datenquelle zu benennen

jp112sdl, 2018-08-19:
Was da schief läuft, kann ich dir so leider auch nicht beantworten.

Seqway, 2018-08-19:
Mhhh,
Dann weiss ich auch nicht mehr weiter.
Ich muss dich nachtraeglich irgendwie die Datenquelle wieder aendern koennen ... gruebelgruebel

jp112sdl, 2018-08-19:
Nur so wie beschrieben. Funktioniert bei mir und bei anderen auch...
Ansonsten hilft dir evtl. noch das serielle Log (115200 Baud) bei der Fehlersuche.

Seqway, 2018-08-19:
Du meinst den seriellen Log beim flashen? Oder kann man auch sehen was der Wemos macht wenn ich darauf zugreifen moechte?

Seqway, 2018-08-19:
Noch was:
Welche Versionen der anderen Biblotheken sind denn zu waehlen? Ich musste ja noch ein paar andere mit einbinden.

jp112sdl, 2018-08-19:
> Du meinst den seriellen Log beim flashen?

Ich meine den seriellen Monitor in der Arduino IDE.
https://www.mymakerstuff.de/2016/04/18/arduino-tutorial-kapitel-13-der-serielle-monitor/
Diesen öffnen, auf 115200 Baud stellen.
Taster an D1 oder D2 gedrückt halten, den kleinen schwarzen Reset-Taster am Wemos drücken (damit der Wemos neu startet) und schauen, was dort angezeigt wird.

> Welche Versionen der anderen Biblotheken sind denn zu waehlen? Ich musste ja noch ein paar andere mit einbinden.

Immer die aktuellen... Ich hab das Projekt irgendwann mal für mich programmiert, hochgeladen und nie weiter gepflegt. Da bei mir alles auf dem aktuellen Stand ist, sollte es aber mit den jeweils neuesten Versionen gehen.

Seqway, 2018-08-20:
Okay werde ich mal probieren !
Ich melde mich dann aber dauert etwas.