jp112sdl/MatrixDisplayESP8266 #17: HTTP Error

dingeberlin, 2018-10-21:
Hi Jerome,
bin neu hier und habe nun alles mit Deiner -dev hinbekommen.
Ich verzweifel nur an der Datenübertrageung. Bekomme immer HTTP Error angezeigt.
Habe ein CuXD Exec Gerät erstellt. ABer scheinbar klappt die kommunikation nicht wie sie soll.
Kannst Du mir bitte noch mal kurz die Code -Schnipsel aus dem ESP config der Url von der CCU schreiben?
Variable in CCU ist als Zeichenkette ist angelegt.
Möglicher Weise aber das Script zum senden an den ESP falsch 

Bist Du so nett und kannst mir helfen?

Danke
Thomas

dingeberlin, 2018-10-21:
Nachtrag:
ESP ist im richtigen Netzwerk
Ping kam mit Antwort zurück.

jp112sdl, 2018-10-21:
Hi!

> Möglicher Weise aber das Script zum senden an den ESP falsch

Es wird nichts an den ESP gesendet. Der ruft nur zyklisch den Variableninhalt ab.

Als häufigste Fehlerquelle fällt mir da zunächst mal die Firewall der CCU ein.
Ist die ESP IP (oder das Netz) für "Remote Homematic-Script API" freigegeben?

dingeberlin, 2018-10-21:
Vielen Dank!!!!!!!!
Daran lag es!!

THX

jp112sdl, 2018-10-21:
Hehe, kein Ding. Viel Spaß damit

closed 2018-10-21 by jp112sdl

dingeberlin, 2018-10-22:
Ich nochmal,

meinst Du man könnte Dein Sketch um eine Sendefunktion erweitern?
Ich würde gern einen Taster an die CCU senden.
Ist Dein WiFi manager dafür grundsätzlich geeignet?
Auf einen Ping antwortet er ja...

Danke

jp112sdl, 2018-10-22:
Hi,

> man könnte 

_Man_ könnte sicherlich alles... Individuelle Modifikationen nehme _ich_ jedoch grundsätzlich nicht vor.

Der WiFiManager ist nur dazu da, um Konfigurationsdaten bequem und dynamisch zu hinterlegen.
Im Normalbetrieb ist der nicht aktiv.
Ist auch nicht *mein* WiFiManager, sondern dieser hier: https://github.com/tzapu/WiFiManager