jp112sdl/MatrixDisplayESP8266 #20: IoBroker - mehr als drei Werte

Seqway, 2018-12-28:
Hallo,
habe das MatrixDisplay mit IoBroker laufen.
Nun wollte ich mehr als drei Werte laufen anzeigen lassen.
Sobald ich den vierten Wert eintrage dann macht das Display nur noch "Starte ..." und switcht zur Uhrzeit. Gehen nur drei Werte ?

jp112sdl, 2018-12-29:
Hi,

es gibt zwar eine Begrenzung, aber erst bei 20.

Evtl. hängt irgendein Sonderzeichen quer?
Schau mal was an seriellem Debug (115220 Baud) ausgegeben wird.

Seqway, 2018-12-29:
gibt es eine beschränkung bei den Zeichen ? ich blende nämlich im IoB noch Text mit ein damit ich weiss was da eingeblendet wird zB Aussentemperatur: Wert°C

jp112sdl, 2018-12-29:
Hast du das °-Zeichen in $ umgewandelt?

jp112sdl, 2018-12-29:
Wegen UTF-8 gibt es ein paar [Beschränkungen](https://github.com/jp112sdl/MatrixDisplayESP8266/blob/a344fc335026e57147b78add99773f70e6222dbb/MatrixDisplayESP8266.ino#L79). Aber eher in der Darstellung... also würdest dann eher falsche Zeichen angezeigt bekommen, anstatt gar nix.

Aber ohne serielles Log können wir da wohl nur raten.

Seqway, 2018-12-29:
Also im iobroker habe ich das zeichen entsprechend umgewandelt. Dort steht --> $C / 
Damit kommt die Temp richtig.
Jetzt kopiere ich den gesamten Baustein um einen weiteren Wert anzeigen zu lassen - von daher müsste die Syntax ja richtig sein aber so bald ich das aktiviere --> "Starte...."

Wie kann ich das denn testen ? einfach anstatt Stromstecker and den Wemos dann an den rechner per USB Kabel und in das ser Log schauen ? Über welches Programm ?

jp112sdl, 2018-12-29:
>Dort steht --> $C / 

Wo kommt der `/` her ? Trennzeichen ist `;`
Siehe Readme:
`"Text1;Text2;Text3;Textn"`

Hab gerade noch mal bei mir geschaut... ich hab 5 Werte drin, das geht problemlos.

> Über welches Programm ? 

z.B. Arduino IDE; Werkzeuge->serieller Monitor (115200 Baud dort einstellen)
Der richtige COM-Port muss natürlich vorher eingestellt werden (Werkzeuge->Port).
Wemos anschließen, seriellen Monitor öffnen, Reset-Taster am Wemos drücken, Ausgaben beobachten.

 - oder - Putty, Serial auswählen, richtigen COM eintragen und 115200 Baud

Seqway, 2018-12-29:
bei mir sieht das so aus im Blockly:

[image]

jp112sdl, 2018-12-29:
Sorry... kenne Blockly nicht.

Was kommt denn im Browser, wenn du die Abruf-URL eingibst?
`http://<CCU-IP>:8181/a.exe?ret=dom.GetObject(%22<VARIABLENNAME>%22).State()`

Seqway, 2018-12-29:
Also, gebe ich den Wert richtig in den Browser ein kommt korrekt in den Datenpunkt der anzuzeigende Text:

"Innentemperatur: 20.4$C  /  PV-Prod Jahr: 1385kWh / PV-Prod aktuell: 0kW Innentemperatur: 20.4$C / "

Mittlerweile geht das Display einfach aus wenn es angezeigt werden soll !!!

jp112sdl, 2018-12-29:
Dein Fehler ist das falsche Trennzeichen... 

Es muss so aussehen:
`"Innentemperatur: 20.4$C;PV-Prod Jahr: 1385kWh;PV-Prod aktuell: 0kW Innentemperatur: 20.4$C"`

_P.S.: Hinten darf kein `;` mehr dran._

Seqway, 2018-12-29:
UNFASSBAR - besser als ich dachte !!!

Genau das wars. und nun wird sogar nur EIN WERT angezeigt - dann bleibt er stehen und dann kommt der nächste ! VORHER war das komplett eine Laufschrift mit ALLEN Werten. GENIAL DANKE !!!!!!!

Seqway, 2018-12-29:
zur Info:

"Innentemperatur: 20.3$C  ;  PV-Prod Jahr: 1385kWh ; PV-Prod aktuell: 0kW ; Innentemperatur: 20.3$C"

jp112sdl, 2018-12-29:
Wenn du noch die Leerzeichen vor und nach dem `;` entfernst, ist es perfekt.

Seqway, 2018-12-29:
bitte schön: :-)

"Aussentemperatur: 6.5$C;Innentemperatur: 20.3$C;PV-Prod Jahr: 1385kWh;PV-Prod Tag: 1.2kWh;PV-Prod aktuell: 0kW"

jp112sdl, 2018-12-29:
So kannst es lassen 👍

closed 2018-12-29 by jp112sdl

Seqway, 2018-12-29:
Super danke nochmal ! Perfekt ...