jp112sdl/MatrixDisplayESP8266 #21: keine daten abgeholt

liv-in-sky, 2019-01-11:
hallo, ich denke, alte projekte zu supporten nervt wahrscheinlich - aber ich würde deine hilfe brauchen. ich habe den d1 mini geflasht bekommen und kann ihn mit web einrichten. die schalter machen auch, was sie sollen - leider wird mein iobroker text nicht abgeholt bzw. nicht angezeigt - ping funktioniert - also im netz angemeldet.
wenn ich im iobroker den simple-api deaktiviere hat das auch keine änderung zu folge - irgendwie startet das das http get nicht? 
meine url sieht so aus: http://192.168.178.59:8087/getPlainValue/controll-own.0.Matrix sieht so aus
mit tasker oder browser kann ich den wert auslesen?
hast du irgendeine idee dazu?

jp112sdl, 2019-01-11:
Hi,

>  irgendwie startet das das http get nicht?

Am besten mal schauen, was seriell ausgegeben wird (115200 baud).

> mit tasker oder browser kann ich den wert auslesen?

Wie sieht denn die Ausgabe konkret aus?

liv-in-sky, 2019-01-11:
hi - estmal danke für deine antwort

momentan habe ich eine uhr - entweder stillstehnede uhrzeit oder links kommend und nach oben verschwindend - mit taster geschaltet

das mit dem seriellen monitor check ich nicht ganz - wenn ich den aufrufe (com8 bei mir), kann ich nichts sehen - der monitor geht auf, das war es - muss ich da noch was machen?
 - im browser ist die ausgabe je nach dem, was ich im iobroker-text-feld" eingebe: bei test ist im browser "test" zu sehen bei "test" im iobroker kommt  "\"test\"" heraus.

ich habe mal anstatt der uhrzeit etwas anderes ausgeben lassen - funktioniert auch - habe die .ino datei geändert um zu sehen, ob die flases auch richtig ankommmen.

liv-in-sky, 2019-01-11:
neuigkeiten beim seriellen monitor - beim schalten mit D2 kommt folgendes - sonst passiert nichts
[image]

liv-in-sky, 2019-01-11:
I'm sorry - habe fehler gefunden - ich hatte wohl den falschen wemos konfiguriert - hatte esp version 2.5 beta2  - der boardtreiber hat einen anderen wemos d1 mini dabei .

werd mich mal im ändern versuchen - hätte gerne eine stehende uhr, wenn http get leer 

und abends würd ich gerne das display ausschalten können - hab sowas noch nie gemacht - werd wohl bis zum frühling dafür brauchen

tolles project - danke für's teilen

closed 2019-01-11 by liv-in-sky